gut-space/satnogs #20: Observation API

tomaszmrugalski, 2020-02-14:
We need a new API that will:

- [ ] receive an observation from reporting ground station
- [ ] rename uploaded file
- [ ] store an observation in a DB
- [ ] rename file using a scheme that ensures no conflicts

It seems that over time more and more parameters will be provided when adding observation. The schema should be easily extensible. One good example would be a POST method with a JSON data and JSON response. This would also ensure easy adaptation by third parties.

tomaszmrugalski, 2020-02-24:
One more thing - the lastobs field in the stations table should be updated.

fivitti, 2020-02-27:
> receive an observation from reporting ground station
We have now API for receive the observations. It accept form-data or JSON formats. It required PNG file with observation and some metadata. After receiving server creates thumbnail.

> rename uploaded file
> rename file using a scheme that ensures no conflicts
Received file is renamed. We merge original filename with UUID (version 4). It allow us to avoid conflicts.
 
> store an observation in a DB
Received observation is inserted to existing DB shema.

fivitti, 2020-03-04:
>  lastobs field in the stations table should be updated

Ready.

tomaszmrugalski, 2020-03-20:
This seems to be complete. Resolving.

closed 2020-03-20 by tomaszmrugalski