gut-space/satnogs #22: authorization, some form of limits

tomaszmrugalski, 2020-02-14:
For the time being we assume a good will of the participants (_**cough**_). However, in the longer term we'll need to have some form of authentication (so we know who is uploading), authentication (only station owner is allowed to report observations of his own stations) and limits (we don't want to turn this into free hosting service).

fivitti, 2020-02-27:
The main tasks:

- [ ] Station authorization
- [ ] User authorization
- [ ] Receiving file limits
- [ ] Max request in specific time limit
- [ ] Max unauthorized requests limit

fivitti, 2020-02-27:
We designed the authorization for ground station.

First we generate cryptographic-safe secret and send it and station ID to station owner.  
Owner provides the secret and ID in station configuration.  
Each observation includes Authorization HTTP header with:

* Timestamp, date when observation is sending
* Station ID
* HMAC signature of observation which is created from:
  * Timestamp
  * Station ID
  * All observation metadata as key-value
  * SHA-1 hash of attached files

After receiving observation server timestamp is current (5 minutes window) and verify signature.  
If it is OK then save observation, otherwise return 401 code.

fivitti, 2020-02-27:
Maybe we should replace SHA-1 with stronger hash for file checksum? Use SHA-1 allows for attacker replace file content.

tomaszmrugalski, 2020-04-11:
User authorization is implemented in #65, currently in review.