gut-space/satnogs #32: DB migrations

tomaszmrugalski, 2020-02-16:
Our current approach to migrations (see backend/db) is a bit primitive. Consider using something similar to: https://pypi.org/project/yoyo-migrations/ or https://pypi.org/project/raw-sql-migrate/

fivitti, 2020-02-16:
Few useful links:

* [Raw SQL migration - Python package](https://pypi.org/project/raw-sql-migrate/)
* [Yoyo migrations - Python package](https://pypi.org/project/yoyo-migrations/)
* [Simply migration script](https://gist.github.com/galvez/607824)  - based on Fabric and sqlwitch, but we can change it for use another background.

fivitti, 2020-03-04:
I prepared migration script - ```migrate-db.py```. It uses ```schema``` table to recognize DB version.
Migration files should be in ```db``` directory in ```.psql```. You need to use name convention:
```
satnogs-XX.psql
```
where XX is schema version. Solution isn't limited to range 00-99. It means that you can use ```satnogs-9999.psql``` and it will work correctly.

In migration script you need to include query for increment (set) ```version``` column in ```schema```. In this way you have a possibility to perform migration manually without Python script.

tomaszmrugalski, 2020-03-08:
Thanks for writing this. I've tested that script and it works well. I think this can be closed now.

closed 2020-03-08 by tomaszmrugalski