gut-space/satnogs #33: Experiment with LRPT reception and decoding

tomaszmrugalski, 2020-02-17:
Right now we're only able to decode APT transmission, which is a 60s technology being used by only 3 NOAA sats and it is being retired. Newer sats (e.g. NOAA-20) no longer use APT. Our current hardware should allow us to receive LRPT transmissions.

According to [wikipedia](https://en.wikipedia.org/wiki/Low-rate_picture_transmission), Metop, NPOESS and [Metor 2-M](https://satsignal.eu/software/LRPT-processor.html) use LRPT.

fivitti, 2020-03-22:
Decoding Meteor-M 2 with QPSK modulation is ready.

We receive signal on 137.1MHz and next demodulate it using [medet](https://github.com/dbdexter-dev/meteor_demod) and next retrieve imagery using [meteor-decoder](https://github.com/artlav/meteor_decoder). We compose RGB image from channels 68, 65 and 64 channels.

I notice that QPSK requires better signal than APT. If maximal elevation is less then 20 degrees then nothing is decoded.

fivitti, 2020-03-22:
Next step: OQPSK modulation in Meteor-M2 2.

fivitti, 2020-03-28:
137MHz LRPT transmission is no longer available.
I stop works on METEOR-M2 2 recipe.

[Source](https://www.rtl-sdr.com/meteor-m-n2-2-weather-satellite-updates-no-more-137-mhz-lrpt-l-x-band-working-in-daylight/)

closed 2020-03-28 by fivitti