gut-space/satnogs #36: Deployment model

fivitti, 2020-02-27:
Deployment model (how to deploy new versions)

After accept changes the new version of software should be automatically build and install on our ground station and content server. Other ground stations connected to content server should get notification that update is required.
All necessary external dependencies must be installed automatically.

fivitti, 2020-03-04:
We have now database migration script - ```migrate-db.py```. It should be included in deployment model.

tomaszmrugalski, 2020-03-26:
I've updated the title. There are two aspects: server upgrades and station upgrades. Let's scope this ticket to server, while the other one (#52) to the station.

tomaszmrugalski, 2020-03-30:
I managed to deploy new server code, but it was tricky as hell.

Here are the issues I had to deal with:

1. satnogs code installs into \~/devel/satnogs/server/venv/lib/python3.7/site-packages/satnogs_server-1.0-py3.7.egg/ and for some reason wsgi runs from there. The installation to site-packages is missing all templates and satnogs.ini is located in a different location (~/devel/satnogs/server/satnogs.ini). I worked around the problem by deleting the ~/devel/satnogs/server/venv/lib/python3.7/site-packages/satnogs_server-1.0-py3.7.egg dir.

2. Following the instructions in the install.md the virtualenv creation was a failure. Here's how it failed and how I fixed it:
```
$ python3 -m virtualenv venv
The path python2 (from --python=python2) does not exist

python3 -m virtualenv --python=python3 venv
```

3. After doing python setup.py install, got the following error:

Finished processing dependencies for satnogs-server==1.0
```
Traceback (most recent call last):
  File "setup.py", line 25, in <module>
    from migrate_db import *
  File "/home/satnogs/devel/satnogs/server/migrate_db.py", line 7, in <module>
    from app.repository import Repository
  File "/home/satnogs/devel/satnogs/server/app/__init__.py", line 7, in <module>
    from flask import Flask
ModuleNotFoundError: No module named 'flask'
```
The problem is the deps were installed, but they're not yet visible in this python interpreter. Rerunning `python setup.py install` works ok.

4. Minor thing: there was leftover cron job:

```
0,10,20,30,40,50 * * * * cd /home/satnogs/devel/satnogs/tools/thumbs.py
```

5. Our setup didn't have python logging configured. I've configure it, so logs from using
standard pythong logging are written to \~/logs/wsgi.log. Here's how you can log something:

```
logging.info("server/app/__init__py: root_dir=%s" % root_dir)
```

tomaszmrugalski, 2020-03-31:
Ok, the server is now upgraded. If things went according to plan, first automated update cycle should be done. The page is up, but the observation files are missing. Need to investigate what's wrong.

tomaszmrugalski, 2020-04-03:
After merging #56 and #59, the upgrade should work now. Resolving.

closed 2020-04-03 by tomaszmrugalski