gut-space/satnogs #39: master doesn't work on python 3.7 and 3.6

tomaszmrugalski, 2020-03-07:
The recent change in backend/app/repository.py imports TypedDict from typing. Unfortunately, on python 3.7.5 it does fails:
```
File "/home/thomson/devel/satnogs/backend/app/repository.py", line 9, in <module>
    from typing import Any, Dict, List, TypedDict, NewType, Sequence, NoReturn, Union, Optional, Tuple
ImportError: cannot import name 'TypedDict' from 'typing' (/usr/lib/python3.7/typing.py)
```

See https://docs.python.org/3.7/library/typing.html and https://docs.python.org/3.8/library/typing.html

fivitti, 2020-03-08:
TypedDict (and Literal) is a part of Python from 3.8 version.

Previously it was a part of [MyPy extensions](https://pypi.org/project/mypy-extensions/). We can import TypedDict from this package when we recognize that Python version is less then 3.8.

We need to analyze which versions of Python we want to support. The content server uses Python 3.8, but Raspberry Pi uses 3.7 (latest version in stable Debian repo).

fivitti, 2020-03-08:
MyPy documentation recommend using [typing_extensions](https://pypi.org/project/typing-extensions/) for TypedDict and Literal.

fivitti, 2020-03-08:
```python
if sys.version_info[0] == 3 and sys.version_info[1] >= 8:
    from typing import TypedDict, Literal
else:
    from typing_extensions import TypedDict, Literal
```

tomaszmrugalski, 2020-03-08:
I think it's time to fix the travis CI tests and start paying attention to their status. It would have been picked by travis (builds on python 3.6, 3.7 and 3.8), but we didn't have the postgres setup completed there. I try to fix that.

tomaszmrugalski, 2020-03-08:
This has been fixed with #40 merge.

closed 2020-03-08 by tomaszmrugalski

Sushang-Agnihotri, 2021-07-16:
Try to Uninstall the related librarie(s) and install it again. 
it worked for me.