gut-space/satnogs #66: pagination exception when clicking last page

tomaszmrugalski, 2020-04-08:
I managed to get this exception when browsing observations list:
```
ERROR:app:Exception on /obslist [GET]
Traceback (most recent call last):
  File "/home/satnogs/devel/satnogs/server/venv/lib/python3.7/site-packages/Flask-1.1.1-py3.7.egg/flask/app.py", line 2446, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/satnogs/devel/satnogs/server/venv/lib/python3.7/site-packages/Flask-1.1.1-py3.7.egg/flask/app.py", line 1951, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/satnogs/devel/satnogs/server/venv/lib/python3.7/site-packages/Flask-1.1.1-py3.7.egg/flask/app.py", line 1820, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/satnogs/devel/satnogs/server/venv/lib/python3.7/site-packages/Flask-1.1.1-py3.7.egg/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/satnogs/devel/satnogs/server/venv/lib/python3.7/site-packages/Flask-1.1.1-py3.7.egg/flask/app.py", line 1949, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/satnogs/devel/satnogs/server/venv/lib/python3.7/site-packages/Flask-1.1.1-py3.7.egg/flask/app.py", line 1935, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/satnogs/devel/satnogs/server/venv/lib/python3.7/site-packages/webargs-5.5.3-py3.7.egg/webargs/core.py", line 455, in wrapper
    return func(*args, **kwargs)
  File "/home/satnogs/devel/satnogs/server/app/pagination.py", line 323, in wrapped
    limit_and_offset=limit_and_offset)
  File "/home/satnogs/devel/satnogs/server/app/controllers/obslist.py", line 15, in obslist
    obs["sat_name"] = satellites_dict[obs["sat_id"]]
KeyError: None
```

Steps to reproduce:
1. go to https://satnogs.klub.com.pl/obslist
2. click on Last page (or any other non-consecutive page, e.g. jump from 1 page to 5th).

You can see the exception on ~satnogs/logs/wsgi.log

fivitti, 2020-04-08:
This problem is fixed in #61 .

fivitti, 2020-04-11:
#61 is merged and this bug is fixed.

closed 2020-04-11 by fivitti